lewisxhe/Esp-badge #1: Poor documentatin

hadmut, 2019-02-04:
Hi, 
I own several of these epaper boards, but are not even sure, which ones. One (2.9, white) says "T5_v2.2 20181116", while the other (seems to be a 2.13 inch epaper board, looks like the one on the picture bottom left, but with four blue LEDs) does not even tell what it is. 

No clue about what the LEDs mean or how to use them, how to use the speaker, pin layout of the connector, or just even how to program them. The arduino IDE claims to program them, but nothing happens. No hint about which ESP32 board to select. 

While the board themselves seem to be technically good, the documentation is poor (worse: non-existing) and renders them more or less useless.

lewisxhe, 2019-02-07:
> Hi,
> I own several of these epaper boards, but are not even sure, which ones. One (2.9, white) says "T5_v2.2 20181116", while the other (seems to be a 2.13 inch epaper board, looks like the one on [...]

Hi hadmut,
I have already noticed that the information is incomplete. I am working hard to ask the team to improve it. At present, the repository I submitted contains pin descriptions and schematics. The library of lilygo was too messy. Now I am trying to perfect it. . You said the four led lights, like the lower left corner is the old version of the T5-V2.0, the four led are just the battery charging indicator, you can remove it if you don't need it.

closed 2019-02-15 by lewisxhe

nano111, 2019-05-14:
Pinout on the board is wrong for version T5v2.2.
I hav found 3 pinsdifferent,dont hav time to look further.
[image]

haldi4803, 2021-04-03:
Taken from the Schematic.... which is probably wrong too... because Ground and 3.3V are correct.
[image]