lewisxhe/Esp-badge #4: T5 V2.1 battery voltage

rlightner, 2019-07-01:
How can I determine current battery voltage?

rlightner, 2019-07-01:
A3 - 4095
A4 - 698
A5 - 264
A6 - 136
A7 - 2240

I'm assuming its A7, but with a full 3.7v lipo don't know what 2240 actually means

rlightner, 2019-07-16:
Solved this, pin 35 has the voltage on it.

closed 2019-07-16 by rlightner

Anton2k, 2021-01-04:
Hey! looking at my T5 v2.3 and I can't see any pin 35, any ideas? trying to measure the voltage of the LiPo thanks. @rlightner

lewisxhe, 2021-01-04:
@Anton2k  
This project is a long time ago, I did not maintain it, the schematic is in this warehouse

https://github.com/Xinyuan-LilyGO/LilyGo-T5-ink-series.git

Anton2k, 2021-01-04:
Thank's @lewisxhe i'll take a look at it, trun's out there is a BAT_TEST pin (35) on my board, no where I though it would be, closer to the bottom of the board, voltage is reading 2000+ though, so I may need to run it through some kind of calculation.

lewisxhe, 2021-01-08:
I have voltage check codes in many examples of lilygo, they are basically the same, like this! !

···
int vref = 1100;

void setup(){
Serial.begin(115200);
esp_adc_cal_characteristics_t adc_chars;
    esp_adc_cal_value_t val_type = esp_adc_cal_characterize((adc_unit_t)ADC_UNIT_1, (adc_atten_t)ADC1_CHANNEL_6, (adc_bits_width_t)ADC_WIDTH_BIT_12, 1100, &adc_chars);
    //Check type of calibration value used to characterize ADC
    if (val_type == ESP_ADC_CAL_VAL_EFUSE_VREF) {
        Serial.printf("eFuse Vref:%u mV", adc_chars.vref);
        vref = adc_chars.vref;
    } else if (val_type == ESP_ADC_CAL_VAL_EFUSE_TP) {
        Serial.printf("Two Point --> coeff_a:%umV coeff_b:%umV\n", adc_chars.coeff_a, adc_chars.coeff_b);
    } else {
        Serial.println("Default Vref: 1100mV");
    }

}

void loop(){

        uint16_t v = analogRead(35);
        float battery_voltage = ((float)v / 4095.0) * 2.0 * 3.3 * (vref / 1000.0);

      Serial.println(battery_voltage );
}
···