lewisxhe/Esp-badge #5: Deep Sleep issues with v2.1 IP5306 power management chip

rlightner, 2019-07-16:
v2.1 (2.7" display) has issues coming out of deep sleep due to the PM IC IP5306 chip. Has anyone solved this issue? Is there a different version of the board that has a 2.7" display that property deep sleeps and wakes?

https://www.rogerclark.net/ttgo-t5-eink-display-update/

jonhassall, 2019-10-24:
Did you find any further information on this? I am encountering a similar issue.

rlightner, 2020-02-16:
Sadly no, still having the issue.

rlightner, 2020-02-19:
@lewisxhe Any chance you can comment on this?

lewisxhe, 2020-02-19:
Sorry, I just saw that I have n’t maintained this project for a long time. The IP5306 you said will automatically shut down, causing esp32 to fail to wake up in deep sleep. This problem is a known issue. You can operate the 5360 through I2C and let it Prohibit light load shutdown. Like here. https://github.com/lewisxhe/TTGO-EPaper-Series/blob/3dcb6469715021d1a3c168ad67025b8128af692d/TTGO-EPaper-Series.ino#L745

rlightner, 2020-02-19:
@lewisxhe  Thanks!
Do you know if they have moved to a different chip in later revisions of the T5 board by chance?

lewisxhe, 2020-02-19:
No, it seems that there is not a good power solution now. 
@rlightner

lyusupov, 2020-03-12:
As far as  I understand, LilyGO is no longer selling V2.1
It was superseeded by V2.8 . Schematics is [there](https://github.com/lewisxhe/TTGO-EPaper-Series/blob/master/schematic/T5_v2.8_27.pdf)
V2.8 is no longer using IP5306

[image]

Pros:
- I have not read IP4054 specs yet but I expect that there is no 50mA threshold with this IC
 So ESP32's 'deep sleep' should be fine.

Cons:
- rightmost key does no longer have ON/OFF function. It would be nice to have a hardware power switch on T5S, similar to other T5-series boards have