lewisxhe/Esp-badge #11: is there a simple working example

fke222, 2020-02-02:
I tried to run the sketch on a LILYGO® TTGO T5 V2.3 2,13" ESP32 with ePaper but it didn't work. 
"cannot declare variable 'display' to be of abstract type 'GxGDE0213B72B'" was the last error.
I have this display V 2.0 and this works fine with <GxEPD2_BW.h><GxEPD2_3C.h> and <U8g2_for_Adafruit_GFX.h> but the sketches for this Display wont work work on V 2.3 items.
My question is - is there a simple example to use this display (V2.3)- lets say print a single string on the ePaper?

lewisxhe, 2020-02-26:
Please note that I am using a fork ， https://github.com/lewisxhe/GxEPD

fke222, 2020-02-26:
Thanks for the Answer. In the meantime I found a working solution. 
I use the "Library: https://github.com/ZinggJM/GxEPD2" with GxEPD2_213_B73.h for the V 2.3 diplay

closed 2020-02-26 by fke222