lewisxhe/Esp-badge #16: T5 V2.3_2.13 Schematic

jpliew, 2020-07-13:
Hi @lewisxhe ,

Could you upload the schematic for T5 V2.3_2.13 please ?

[image]

Thanks

ejs-ejs, 2020-10-24:
This board uses GxGDEH0213B73.h EPD.
Pin connections seem to match v2.3 board

jpliew, 2020-10-25:
@ejs-ejs thanks for your reply. They are not completely the same. This board uses IO19 for LED, but v2.3 uses IO22

ejs-ejs, 2020-10-25:
So, boards_defs.h can look like this?
I've saved the file as librarys, and now i can simply
`#define TTGO_T5_2_3 213 // T5 2.3 rev 2.13
#include <TTGO_T5_boards.h>`

`
...
// TTGO T5 2.0 or 2.3
#elif (TTGO_T5_2_0) || (TTGO_T5_2_3)
 #if TTGO_T5_2_3 == 213
 #pragma message "Including rev 2.13 screen, GxGDEH0213B73.h"
   #include <GxGDEH0213B73/GxGDEH0213B73.h> // rev 2.13
 #else
  // #include <GxGDE0213B1/GxGDE0213B1.h>      // 2.13" b/w
  // #include <GxGDE0213B72/GxGDE0213B72.h>      // 2.13" b/w     GxGDE0213B1 升级版本
  //#include <GxGDEH0213B72/GxGDEH0213B72.h> // 2.13" b/w           GxGDE0213B72 升级版本 默认LilyGO的出厂屏幕都是这种
  #include <GxGDEH0213B72/GxGDEH0213B72.h>
  #pragma message "Screen is GxGDEH0213B72.h"
 #endif 

  #define SPI_MOSI 23
...
  #define SPEAKER_OUT -1
  
  #if TTGO_T5_2_3 == 213
    #pragma message "LED on IO19"
    #define LED 19
  #else 
	#pragma message "LED on IO22"
    #define LED 22
  #endif 
...
  #endif`

jpliew, 2020-10-25:
@ejs-ejs thanks for your reply. I have no problem using or programming the board.

I am actually interested to see the hardware differences between the different versions, hence asking for the schematic.

grahamwhaley, 2020-12-05:
ack - I'm interested too. I'm trying to get it into the lowest power I can in hibernate, and need accurate schematic to help with that. I also noted that the [uploaded schematic](https://github.com/lewisxhe/TTGO-EPaper-Series/blob/master/schematic/T5_V2.3.pdf) has the LED different from the board (my board looks exactly like the one in the photo on this Issue). I note the uploaded schematic says 'T5_2.0' inside it. If we can get or find a link to the correct schematic that would be great.

pls note, this repository is great, and has been a great help - nice work. Just looking to see if we can get an upto date schematic at all.

grahamwhaley, 2020-12-05:
Oh, an addition. My board lights up a blue led when I plug in the battery. The schematic has no blue LED on it. I guess they swapped red LED_G2 for a blue led. 
I wish they had wired that LED to VBUS and not +3v3, as it will kill my hibernate mode current draw ;-)

BetweenBeltSizes, 2020-12-16:
@grahamwhaley @lewisxhe hello, absolute n00b here, just managed to get my 1st esp32 ttgo up and running with a quote machine i followed from a guide on "re-innovation.co.uk", so this Blue LED on the back the only way to turn it off would be removing it? for me its always on whether i power from USB or 3.7V 18650 battery... what about you guys? also how would one measure current draw in deep sleep? putting an ammeter between battery and power input of esp32 is the only way or is there a way to show the power draw in software? Many thanks, sincerely, a n00b

grahamwhaley, 2020-12-16:
Hi @BetweenBeltSizes . As far as I can tell the only way to turn off the blue power LED is to remove it. I removed the actual LED from my board, but it might be better longer term to remove the resistor feeding the LED, as that would be easier to replace later if you ever needed to - R22 on the schematic. However, iirc, the resistors are '0402' size, which is tiny. The LED is '0603' I think, and a little easier to remove. Still a little tricky. I basically ended up melting mine off.

For current draw, there is no built in way to measure that on the board. Yes, you insert an ammeter in the power feed line. I placed mine on the battery + lead. Be aware, measuring very small low voltage currents is a challenge. The ammeter will be using an in-built resistor to measure the current, and the voltage drop across that resistor can be enough to stop your board booting. I can measure OK on one of my (cheap) meters using the 20mA range, but if I use a lower range the voltage drops too much and the board gets stuck in a constant reset cycle.

I have gotten my board down to 460uA idle. Here is a short summary of things to do:

 - remove the power LED
 - Add a pullup on the SPI flash chip select to allow it to enter a lower power mode

From my calculations, I think the theoretical minimum idle power is around 250uA, so I'm still 'losing' 200uA somewhere, and I don't know where :-) But, I've stopped trying to get it lower now.

And then, for the actual running time (when the processor is not sleeping), depending on what your code needs to do, you can drop the CPU clock speed - I boot my board at 80MHz (the lowest you can go for wifi), and then drop it to 20MHz in software after I'm done with the wifi and am updating the screen. Most of the time updating the screen is just waiting for the screen to finish and not be 'busy', so you may as well run the CPU as slow as you can at that point and not burn unnecessary power.

I have a repo with a dual-timezone clock implementation which I'll push up at some point.

BetweenBeltSizes, 2020-12-17:
@grahamwhaley thank you for all the info, i just measured mine when its idle it draws 3.6 mA, when updating the display and fetching info from url over wifi my multimeter showed about 120-150 mA. When mine draws 3.6mA its in deep sleep i think, im not too familiar with esp32 and programming mcus yet, if you could share your github link to your code you use for those uA power consumption at some would be awesome. Pull up to the spi in software? Or do you mean add an actual pull up resistor!? That shit would be hardcore.

grahamwhaley, 2020-12-17:
*Graham checks his notes...*

Hi @BetweenBeltSizes . I did see 3.6mA too in deep sleep on one of my projects. The other project showed 1.9mA initially - I never did figure out why they were different - it might be a build setting.

Physically removing the blue power led will get you a 0.4mA reduction. Physically wiring a pullup (I used 1mohm) on the SPI flash chip chip select will save you another 1mA in deep sleep. Yes, sorry, it is a physical wiring with a soldering iron to some quite small pins. There is no way to disable the led or afaik put the SPI flash into it's first stage sleep mode from sw :-(

My code is not ready to push to github yet - I need to clean it up and the project is not quite finished yet (I'm trying to put a solar panel charger on it to make it standalone and wire free). If you ping me an email (firstname -dot- lastname @ gmail -dot- com), I can fire you some code snippets and a couple of pictures of the modifications.

lewisxhe, 2021-01-04:
This project is a long time ago, I did not maintain it, the schematic is in this warehouse

https://github.com/Xinyuan-LilyGO/LilyGo-T5-ink-series.git